OpenSourceEBike/BMSBattery_S_controllers_firmware #3: hex2bin vs objcopy

spth, 2017-08-10:
The current Makefile uses hex2bin, which does not seem to be a very common utility (e.g. there is apparently no Debian package for it). This makes it a little bit harder for people to compile the firmware.

How about using objcopy (which i part of GNU binutils, and thus easily available on typical GNU/Linux systems) instead?
objcopy --input-target=ihex --output-target=binary file.ihx file.bin should do for converting Intel hex to binary.

Or use makebin, which is included in SDCC (and thus no extra effort to get, since SDCC is needed to compile the firmware anyway)?

Since a lot of software (e.g. stm8flash) can handle .ihx files directly, it might even make sense to use the .ihx file as final output.

Philipp

casainho, 2017-08-10:
Thank you Philipp and in fact, my latest code and Makefile is on the "SVM" branch -- that one already uses objcopy -- It took me a few days to remake the Makefile in a way to build for debug and also a bin file to be flashed, as also to build only the changed C files (unfortunately, can't detect changed header files...).

shoffmeister, 2017-10-18:
I believe this issue is resolved, right?

casainho, 2017-10-18:
Yrs, now we use objcopy.

closed 2017-10-18 by casainho