OpenSourceEBike/BMSBattery_S_controllers_firmware #25: Battery Level

butchja, 2018-01-21:
Just an idea:
Battery level should not display absolute battery voltage. The internal resistance should be comensated so a grater range can be achieved. Firmware should measure the battery voltage in idle and estimate internal resistance under known load conditions (ohms law). Since battery is depleading by time some kind of short intervall might be used for this internal resistance estimation.

casainho, 2018-01-21:
I would like to understand better: "The internal resistance should be compensated so a greater range can be achieved." <--- what is the idea with this issue, have a better battery reading?? or get out the most efficiency from battery?

_Power transfer between a voltage source and an external load is at its most efficient when the resistance of the load matches the internal resistance of the voltage source._
http://farside.ph.utexas.edu/teaching/302l/lectures/node62.html

butchja, 2018-01-21:
I did not express myself clearly. It is not about impedance matching between source and load. As far as I understand this is not applicable here(?). It is more about detecting the real battery level and only then switch off if the battery is really depleded. 
With low battery and in case of load, the raw voltage readings are be below the undervoltage threshold, but this is permissible.

casainho, 2018-01-21:
More ideas about this subject:

https://endless-sphere.com/forums/viewtopic.php?p=1352669#p1352669

Re: Flexible OpenSource firmware for BMSBattery S/Kunteng KT motor controllers (0.25kW up to 5kW)
Report this post Quote
Post  by honya96 » Sun Jan 21, 2018 3:33 pm

It has nothing to do with battery level monitor but with some smart soft cut-off LVC which will lower your max amps when you start getting close to lvc voltage. Even just in theory its not easy because someone can have sag of half the usable voltage range and someone less than a volt with lipo so it will never help him.

Even with Adaptto - a really expensive controller. The soft cut-off didnt work 100% with my weak battery. Under high load it hit the lvc too fast so it just cut the power and applied back. But when not using 7kw but just 1kw down to empty battery, then it started lowering the power so I was able to get home everytime. Even just 100w walking speed at exact lvc voltage. But that way I have never run out of juice totaly, far from home 

For the battery monitor, as I mentioned before, its enough to refresh only when no power applied to motor. So you go with full battery, 20km straight without ever releasing throttle, then wonder "whats my capacity left?" You release throttle, wait a second or less just so the no load voltage stabilizes. And then it changes from full (last remembered value) to 2 bars for example.

I guess it will also use less processing power than when its still jumping depending on load.