OpenSourceEBike/BMSBattery_S_controllers_firmware #29: Street and offroad modes

casainho, 2018-03-11:
For legal purposes in EU, on pedal bike, Throttle can work only to 6km/h.
There are some discussion on the idea to be have two working modes: street and offroad, where on offroad  there would be no limitations. User could switch modes by doing:
- you press and release brake few times in specific intervals and it changes the mode (remembers even after turned of and on - storing a flag on EEPROM); only can switch when at standstill.

casainho, 2018-03-11:
I like that idea of booting always in street mode and thinking that mode changing will happens rarely. But, for users living in places without limitations, a flag on java tool would boot firmware always in offroad mode (no limitations).

cepen, 2018-09-13:
This is one of the best features in this firmware for higher power controllers. One more on my wishlist is ESP